pyobs/pyobs-core #46: Access control

thusser, 2021-12-01:
Would be good to have so some of access control, to restrict access to modules (or even their single methods) to certain other modules. Would probably be nice to have groups as well.

thusser, 2021-12-01:
A simple way to achieve this would be a definition of permissions in the config, like this:

```
acl:
  class: pyobs.acs.BaseACL
  rules:
    - rule: allow
      users: *
    - rule: deny
      groups: [students]
```

The `rules` would be applied one after another. So in this example, the first rule allows access to all, but then denies access to all users in the `students` group.

Another set of rules could be:

```
  rules:
    - rule: deny
      users: all
    - rule: allow
      groups: [admin]
      methods: [move_radec]
```

First, access is denied to everyone, but then users of group `admin` are permitted to call `move_radec` on this module.

So, a rule would be built from up to 4 keywords:
- `rule`: `allow` or `deny`
- `users`: either `all` or a list of usernames (optional, default is `all`)
- `groups`: list of groups (optional)
- `methods`: list of methods (optional)

karlwessel, 2021-12-06:
I don't have much experience with security but this sounds like a lot of work to get this working correctly. 

What would be the specific use cases?

Does XMPP provide some functionality to restrict communication between certain users or groups?

thusser, 2021-12-06:
I think a basic implementation would be quite simple. Modules have a single point of entry for calls from outside (the Comm object), so I can just check the call against a list of permissions there.

The use case would be something like the lab course, where students should not get access to the whole system.

An no, I cannot think of a good way to restrict communication within XMPP. It even seems difficult to have something like groups in the network.

thusser, 2024-09-20:
Moved to Mantis

closed 2024-09-20 by thusser