pyobs/pyobs-core #382: No valid authentication method to ejabberd

jpsbento, 2025-02-06:
I have been trying to run a simple DummyTelescope test using containers, and attempting to connect to an Ejabberd server. 

However, I keep getting authentication errors when the telescope module attempts to connect. 

This is my docker compose file: 

```
services:

  ejabberd:
    image: ejabberd/ecs:latest
    container_name: ejabberd
    volumes:
      - ./database:/opt/ejabberd/database
      - ./ejabberd.yml:/opt/ejabberd/conf/ejabberd.yml:ro
      - ./startup.sh:/opt/ejabberd/startup.sh:ro
    entrypoint: ["/opt/ejabberd/startup.sh"]
    ports:
      - "5222:5222"
      - "5269:5269"
      - "5280:5280"
      - "5443:5443"
    healthcheck:
      test: ["CMD", "ejabberdctl", "status"]
      interval: 10s
      retries: 5

  telescope:
    container_name: telescope
    build:
      context: .
      dockerfile: Dockerfile
    volumes:
      - ./telescope.yaml:/app/pyobs/telescope.yaml:ro
      - ./_environment.yaml:/app/pyobs/_environment.yaml:ro
    depends_on:
      xmppserver:
        condition: service_healthy
    entrypoint: ["pyobs", "telescope.yaml"]

  ```

I then use a simple Dockerfile for the telescope module, which just installs the dependencies for pyobs and pyobs-core and pyobs-gui using pip. 

My telescope.yaml file is quite simple too, but I segregate the domain (localhost) from the server (ejabberd): 

```
{include _environment.yaml}

class: pyobs.modules.telescope.DummyTelescope

comm:
    class: pyobs.comm.xmpp.XmppComm
    jid: telescope@localhost
    password: telescopepass
    server: ejabberd:5222
    use_tls: false
``` 
 
And my startup.sh script for the ejabberd server is: 

```
#!/bin/sh

echo "Starting ejabberd"
ejabberdctl start
echo "Waiting for ejabberd to be ready..."
while ! ejabberdctl status | grep -q "The node ejabberd@localhost is started"; do
  sleep 2
done
echo "ejabberd started"

echo "Creating shared roster group"
ejabberdctl srg_create all localhost all all all
echo "Registering users"
ejabberdctl register telescope localhost telescopepass
ejabberdctl register admin localhost adminpass   

ejabberdctl srg_user_add @all@ localhost all localhost
echo "Users created and added to shared roster"

echo "Querying status of ejabberd"
ejabberdctl status
echo "ejabberd status queried"

echo "Listing shared roster groups"
ejabberdctl srg_list localhost

echo "Tailing logs"
tail -f /opt/ejabberd/logs/ejabberd.log

```
I also change a couple of the settings on the ejabberd.yml configuration, in which I set `auth_method: internal` and disable tls on the port 5222 listener. Other than that, I've kept the defaults. 

Could you please advise if there is anything I'm missing in order to get the telescope to connect to the ejabberd server? The kind of errors I'm getting are: 

2025-02-06 22:09:14,744 [INFO] application.py:133 Opening module...2025-02-06 22:09:14,891 [INFO] mechanisms.py:260 Authentication failed: not-authorized2025-02-06 22:09:14,892 [INFO] client.py:175 SASL: X-OAUTH2: Missing credential: access_token2025-02-06 22:09:14,892 [INFO] client.py:175 SASL: PLAIN: PLAIN without encryption2025-02-06 22:09:14,892 [ERROR] mechanisms.py:188 No appropriate login method.2025-02-06 22:09:14,892 [ERROR] xmppclient.py:76 Invalid credentials for connecting to XMPP server.2025-02-06 22:09:14,894 [INFO] motionstatus.py:76 Changed motion status to MotionStatus.IDLE.2025-02-06 22:09:14,896 [INFO] xmlstream.py:625 connection_lost: (None,)2025-02-06 22:09:14,896 [INFO] xmppcomm.py:232 Disconnected from server, waiting for reconnect...2025-02-06 22:09:16,936 [INFO] mechanisms.py:260 Authentication failed: not-authorized
2025-02-06 22:09:16,936 [INFO] client.py:175 SASL: X-OAUTH2: Missing credential: access_token
2025-02-06 22:09:16,936 [INFO] client.py:175 SASL: PLAIN: PLAIN without encryption
2025-02-06 22:09:16,937 [ERROR] mechanisms.py:188 No appropriate login method.
2025-02-06 22:09:16,937 [ERROR] xmppclient.py:76 Invalid credentials for connecting to XMPP server.
Traceback (most recent call last):
  File "/usr/local/bin/pyobs", line 8, in 
    sys.exit(main())
  File "/usr/local/lib/python3.10/site-packages/pyobs/cli/pyobs.py", line 102, in main
    run(app_class=Application, **args)
  File "/usr/local/lib/python3.10/site-packages/pyobs/cli/pyobs.py", line 85, in run
    app.run()
  File "/usr/local/lib/python3.10/site-packages/pyobs/application.py", line 99, in run
    self._loop.run_until_complete(main)
  File "/usr/local/lib/python3.10/asyncio/base_events.py", line 647, in run_until_complete
    raise RuntimeError('Event loop stopped before Future completed.')
RuntimeError: Event loop stopped before Future completed.

I know the password is set correctly, so that's not the problem, despite the fact that the log seems to suggest that. I suspect it's a problem with the authentication method that pyobs supports. 

Thanks in advance.

thusser, 2025-02-13:
Sorry for the late reply, I'm overseas at the moment. Looks good to me, although I never started pyobs in a container. You can check your ejabberd server by connecting from another client, e.g. Pidgin or just any chat client that supports XMPP. See whether you can connect. I'll try to reproduce the error when I'm back home next week. Could I get your Dockerfile?

jpsbento, 2025-02-13:
Hi! 

Sure, I'll have a go from another client, see if that gives me more of a clue. Thanks for looking into this. 

My dockerfile is this, which you can see if pretty simple. Seems like pyobs-core doesn't install a couple of things, so I had to add them later, but other than that this seems to work fine. 

```
FROM python:3.10-slim-buster

WORKDIR /app/pyobs

RUN pip install pyobs-core pyobs-gui 

RUN pip install pandas 'photutils<1.6' paramiko 

ENTRYPOINT ["pyobs"]
```

jpsbento, 2025-07-09:
Thank you for looking into this! I will try it again soon.